Genesis-Embodied-AI/RoboGen #2: Possibility of changing to another robot arm model

Radiance-nt, 2023-11-05:
Hi, thanks for opening this repository. It's kind of you to share this excellent work. 

I was wondering if it would be possible to integrate a custom robot arm model into the simulated environment (e.g. providing urdf files)? The goal would be to transfer policies trained in simulation to a real robot arm.

Thanks in advance!

zhouxian, 2023-11-05:
Yes it's totally possible. Our different robot types also come from different URDFs. You just need to add them when creating the tasks. Maybe @yufeiwang63 can add a section explaining details for doing so.

yufeiwang63, 2023-11-05:
Yeah it's definitely doable. For manipulation tasks, these three files define the robots we use: [sawyer](https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/sawyer.py), [franka](https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/panda.py), and [ur5](https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/ur5.py). 

As the file indicates, the essential change needed for a new robotic arm would be to change the source urdf file, and the controllable joint indices as well as the finger index of the arm, e.g., for franka the urdf file and joint indices are defined at these lines: [urdf](https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/panda.py#L28), [joint indices](https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/panda.py#L16-L18).

For locomotion tasks, similarly, these [lines](https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/locomotion/sim.py#L140) import the corresponding robot into the simulation scene. Again, what you would need to do is to add a branch for the new robot you have (with the urdf files of the new robot).

Let me know if this helps!

Radiance-nt, 2023-11-06:
> Yes it's totally possible. Our different robot types also come from different URDFs. You just need to add them when creating the tasks. Maybe @yufeiwang63 can add a section explaining details for doing so.

> Yeah it's definitely doable. For manipulation tasks, these three files define the robots we use: [sawyer](https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/sawyer.py), [...]

These suggestions are very helpful, thanks for your kind replies!