Genesis-Embodied-AI/RoboGen #5: ompl isn't getting installed / not findable -> can't run example code

StoneT2000, 2023-11-07:
I got this error while trying to run the example code after running the ompl installation code

```
ython execute.py --task_config_path example_tasks/Change_Lamp_Direction/Change_Lamp_Direction_The_robotic_arm_will_alter_the_lamps_light_direction_by_manipulating_the_lamps_head.yaml
pybullet build time: May 20 2022 19:45:31
all substeps:
  grasp the lamps head
 change the lamps light direction
all substep types:
 primitive
reward
all action spaces:
 None
delta-translation
executing for substep:
 grasp the lamps head primitive
Traceback (most recent call last):
  File "/home/stao/work/external/RoboGen/pybullet_ompl/pb_ompl.py", line 2, in <module>
    from ompl import base as ob
ModuleNotFoundError: No module named 'ompl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/stao/work/external/RoboGen/execute.py", line 272, in <module>
    execute(task_config_path, resume=args.resume, training_algo=args.training_algo, time_string=args.time_string, 
  File "/home/stao/work/external/RoboGen/execute.py", line 192, in execute
    rgbs, states = execute_primitive(task_config_path, solution_path, substep, last_restore_state_file, save_path, 
  File "/home/stao/work/external/RoboGen/execute.py", line 14, in execute_primitive
    env, safe_config = build_up_env(task_config, solution_path, task_name, last_restore_state_file, 
  File "/home/stao/work/external/RoboGen/manipulation/utils.py", line 155, in build_up_env
    module = importlib.import_module("{}.{}".format(solution_path.replace("/", "."), task_name))
  File "/home/stao/mambaforge/envs/robogen/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/stao/work/external/RoboGen/example_tasks/Change_Lamp_Direction/task_Change_Lamp_Direction/grasp_the_lamps_head.py", line 4, in <module>
    from manipulation.gpt_primitive_api import *
  File "/home/stao/work/external/RoboGen/manipulation/gpt_primitive_api.py", line 5, in <module>
    from manipulation.motion_planning_utils import motion_planning
  File "/home/stao/work/external/RoboGen/manipulation/motion_planning_utils.py", line 2, in <module>
    import pybullet_ompl.pb_ompl as pb_ompl
  File "/home/stao/work/external/RoboGen/pybullet_ompl/pb_ompl.py", line 9, in <module>
    from ompl import base as ob
ModuleNotFoundError: No module named 'ompl'
```

Any idea?

StoneT2000, 2023-11-07:
There seems to also be some errors that show up upon trying to install the py bindings for ompl

```
/usr/local/lib/python3.10/dist-packages/pyplusplus/binary_parsers/parsers.py:7: DeprecationWarning: invalid escape sequence 'm'
  """
Traceback (most recent call last):
  File "/home/stao/work/external/RoboGen/ompl-1.5.2/py-bindings/generate_bindings.py", line 950, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/stao/work/external/RoboGen/ompl-1.5.2/py-bindings/generate_bindings.py", line 878, in __init__
    code_generator_t.__init__(self, 'util', None, replacement, 1)
  File "/home/stao/work/external/RoboGen/ompl-1.5.2/py-bindings/ompl/bindings_generator.py", line 146, in __init__
    self.filter_declarations()
  File "/home/stao/work/external/RoboGen/ompl-1.5.2/py-bindings/generate_bindings.py", line 893, in filter_declarations
    self.std_ns.class_('vector< std::string >').include()
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/scopedef.py", line 554, in class_
    recursive=recursive)
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/usr/local/lib/python3.10/dist-packages/pygccxml/declarations/scopedef.py", line 90, in get_single
    raise runtime_errors.declaration_not_found_t(decl_matcher)
```

yufeiwang63, 2023-11-08:
Can you provide more details of your system?
I tested the installation of ompl 1.5.2 on Ubuntu 18.04, with system-wide python version being 3.6.9. 

Another option is for us to provide a docker image that have ompl already installed. We will look into that if installing ompl is too much pain on other systems.

Kira0096, 2023-11-08:
> Can you provide more details of your system? I tested the installation of ompl 1.5.2 on Ubuntu 18.04, with system-wide python version being 3.6.9. The pyplusplus version is 1.8.5. pygccxml version is 2.4.0. Besides, I installed castxml by this (based on my system is 18.04)
> 
> ```
>        wget -q -O- https://data.kitware.com/api/v1/file/5e8b740d2660cbefba944189/download | tar zxf castxml-linux.tar.gz -C ${HOME}
>         export PATH=${HOME}/castxml/bin:${PATH}
> ```
> 
> Let me know if this helps.
> 
> Another option is for us to provide a docker image that have ompl & conda env already installed. We will look into that if installing ompl is too much pain on other systems.

give us a docker image pls!!!

yufeiwang63, 2023-11-09:
I will look into building a docker image and update the repo once it's done.

yufeiwang63, 2024-03-20:
Hi,

If you still have trouble installing ompl python binding from source, you can try installing it from the pre-built python wheels, see the updated ReadMe for details.

closed 2024-03-27 by yufeiwang63