Genesis-Embodied-AI/RoboGen #8: ompl.util._util error

qiaojunyu, 2023-11-10:
python execute.py --task_config_path example_tasks/Change_Lamp_Direction/Change_Lamp_Direction_The_robotic_arm_will_alter_the_lamps_light_direction_by_manipulating_the_lamps_head.yaml  

 grasp the lamps head primitive
Traceback (most recent call last):
  File "/ssd/code/RoboGen/pybullet_ompl/pb_ompl.py", line 2, in <module>
    from ompl import base as ob
  File "/home/qiaojun/miniconda3/envs/robogen/lib/python3.9/site-packages/ompl/base/__init__.py", line 1, in <module>
    from ompl import util
  File "/home/qiaojun/miniconda3/envs/robogen/lib/python3.9/site-packages/ompl/util/__init__.py", line 4, in <module>
    from ompl.util._util import *
ModuleNotFoundError: No module named 'ompl.util._util'

ubuntu 20.04

yufeiwang63, 2023-11-10:
That means ompl is not correctly installed. Could you please take a look at https://github.com/ompl/ompl/issues/609.
We will also provide a docker with ompl installed.

Radiance-nt, 2023-11-12:
OMPL py-bindings installation errors. You must have `_utils.so` and other libs generated under `build/Release/lib`, you can refer to [ompl](https://blog.csdn.net/qq_38606680/article/details/129028660).

GreenAvocado92, 2023-11-14:
> hi, great work python execute.py --task_config_path example_tasks/Change_Lamp_Direction/Change_Lamp_Direction_The_robotic_arm_will_alter_the_lamps_light_direction_by_manipulating_the_lamps_head.yaml
> 
> grasp the lamps head primitive Traceback (most recent call last): File "/ssd/code/RoboGen/pybullet_ompl/pb_ompl.py", line 2, in from ompl import base as ob File "/home/qiaojun/miniconda3/envs/robogen/lib/python3.9/site-packages/ompl/base/**init**.py", line 1, in from ompl import util File "/home/qiaojun/miniconda3/envs/robogen/lib/python3.9/site-packages/ompl/util/**init**.py", line 4, in from ompl.util._util import * ModuleNotFoundError: No module named 'ompl.util._util'
> 
> ubuntu 20.04

Hi，
Did you solve this problem？

Radiance-nt, 2023-11-14:
> Hi， Did you solve this problem？

I have solved that problem on ubuntu20.04. It was hard. I recommand creating a virtual python env instead of using `/usr/bin/python` to compile the py-bindings. Install `numpy`, ` pygccxml==2.2.1` and `pyplusplus` in that venv before Cmaking. And then examine if there is any error in `making update_bindings`. If all succeeds, ` _utils.so` and other libs will be generated under `build/Release/lib`.

GreenAvocado92, 2023-11-14:
> I have solved that problem on ubuntu20.04. It was hard. I recommand creating a virtual python env instead of using `/usr/bin/python` to compile the py-bindings. Install `numpy`, ` pygccxml==2.2.1` [...]

I have installed pygccxml&& pyplusplus, but when I excute install_ompl_ubuntu.sh,  report errors as:

-- The following Python modules were NOT found:
 * pyplusplus pygccxml

Radiance-nt, 2023-11-14:
> I have installed pygccxml&& pyplusplus, but when I excute install_ompl_ubuntu.sh, report errors as:
>
> -- The following Python modules were NOT found:
>
>     * pyplusplus pygccxml

Do you have the required python libraries installed correctly in the python environment used by cMake? You can specify the python for cmake with param  ` -DPYTHON_EXECUTABLE <your path to python>`
And, by the way, you should better compile OMPL mannually rather than run that script. Try to understand the procedure `install_ompl_ubuntu.sh`, it is quite simple. You may need to adjust the cmake configurations.

koupoo, 2023-11-15:
> hi, great work python execute.py --task_config_path [...]

Encountered the same problem and took some time to solve it, and found that the version of pypy3 was incorrect , just upgrage pypy3 and compile ompl again:
```
sudo apt-get remove pypy3
sudo add-apt-repository ppa:pypy/ppa
sudo apt-get install pypy3
sudo apt update sudo apt install pypy3
pypy3 -V
Python 3.9.18 (7.3.13+dfsg-1~ppa1~ubuntu20.04, Oct 17 2023, 11:00:05)
[PyPy 7.3.13 with GCC 9.4.0]
```

hilookas, 2023-12-09:
I solved this problem by using latest ompl source from github, following this approach:

in `install_ompl_1.5.2.sh` comment the last line:
install_ompl -> #install_ompl

Run `install_ompl_1.5.2.sh` to install apt requirements.

Update my pypy3 install following @koupoo 's guide.

Compile ompl with guide from Github

`git clone https://github.com/ompl/ompl.git`

```bash
mkdir -p build/Release
cd build/Release
cmake ../..
# next step is optional
make -j 4 update_bindings # if you want Python bindings
make -j 4 # replace "4" with the number of cores on your machine
```

Don't forget to add py binding path to {your_conda_install}/envs/robogen/lib/python3.9/site-packages/ompl.pth

For my installation, py binding path is `/home/ubuntu/RoboGen/ompl/py-bindings`

DavoodSZ1993, 2024-01-27:
> I solved this problem by using the latest ompl source from github, following this approach:
> 
> in `install_ompl_1.5.2.sh` comment out the last line: install_ompl -> #install_ompl
> 
> Run `install_ompl_1.5.2.sh` to install apt requirements.
> 
> Update my pypy3 install following @koupoo 's guide.
> 
> Compile ompl with the guide from Github
> 
> `git clone https://github.com/ompl/ompl.git`
> 
> ```shell
> mkdir -p build/Release
> cd build/Release
> cmake ../..
> # next step is optional
> make -j 4 update_bindings # if you want Python bindings
> make -j 4 # replace "4" with the number of cores on your machine
> ```
> 
> Don't forget to add py binding path to {your_conda_install}/envs/robogen/lib/python3.9/site-packages/ompl.pth
> 
> For my installation, py binding path is `/home/ubuntu/RoboGen/ompl/py-bindings`

Hey, 
I exactly followed your steps for ubuntu 20.04, but I get the following error.
[ 74%] Built target test_nearestneighbors
/usr/bin/ld: CMakeFiles/test_2dpath_simplifying.dir/geometric/2d/2dpath_simplifying.cpp.o: in function `SimplifyTest::SimplifyTest()':
2dpath_simplifying.cpp:(.text._ZN12SimplifyTestC2Ev[_ZN12SimplifyTestC5Ev]+0x1cc): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
/usr/bin/ld: 2dpath_simplifying.cpp:(.text._ZN12SimplifyTestC2Ev[_ZN12SimplifyTestC5Ev]+0x819): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [tests/CMakeFiles/test_2dpath_simplifying.dir/build.make:93: bin/test_2dpath_simplifying] Error 1
make[1]: *** [CMakeFiles/Makefile2:1124: tests/CMakeFiles/test_2dpath_simplifying.dir/all] Error 2
[ 74%] Linking CXX shared module ../lib/_util.so
Copying python module util into place
[ 74%] Built target py_ompl_util
make: *** [Makefile:163: all] Error 2
davood@davood-Latitude-3440:~/Downloads

Any Ideas?
Thanks

zkingston, 2024-01-30:
@yufeiwang63

Hi, I'm the maintainer of OMPL. If possible, I'd recommend using the pre-built Python wheels (available here currently: https://github.com/ompl/ompl/releases/tag/prerelease). We plan to release on PyPI soon. 

Happy to help with build issues - I'd recommend if possible using the latest version of OMPL, as 1.5.2 has some known issues with compilation.

DavoodSZ1993, 2024-01-31:
Thanks,

I clone the latest version from your Github repo. However, I still have a problem with `boost`. I already have raised an issue regarding this.

I have no idea how should I integrate this Python wheels with OMPL, as I'm new in this area. 

I wonder whether is there any other way to use OMPL Python bindings on other platforms (Windows)?

Thanks again

GenerativeSimulation, 2024-02-03:
> @yufeiwang63
>
> Hi, I'm the maintainer of OMPL. If possible, I'd recommend using the pre-built Python wheels (available here currently: https://github.com/ompl/ompl/releases/tag/prerelease). We plan [...]

Thanks! I used ompl 1.5.2 because I had trouble building the latest github source (I think it's version 1.6) back then. 
It would be great to see you are planning to release the prebuild wheel! I will update the readme once that's stable, e.g., released on PyPI. I will also update the readme with the current prebuilt wheel link you posted. 

Thanks for looking into this issue!

yufeiwang63, 2024-02-03:
> @yufeiwang63
>
> Hi, I'm the maintainer of OMPL. If possible, I'd recommend using the pre-built Python wheels (available here currently: https://github.com/ompl/ompl/releases/tag/prerelease). We plan [...]

Thanks! I used OMPL 1.5.2 because I had trouble building from the latest source back then (I think it was version 1.6). 
It's great to know that you have the prebuilt python wheels! I have updated the readme with your suggestion. 

Thanks for looking into this issue!

yufeiwang63, 2024-03-20:
Hi,

If any of you are still having trouble installing ompl python binding from source, please take a look at installing it from the pre-built python wheels, see the updated ReadMe for details.

closed 2024-03-27 by yufeiwang63