Genesis-Embodied-AI/RoboGen #14: cannot download the dataset

Guanbin-Huang, 2023-12-19:
[image]

file is not available in the site
[image]

yufeiwang63, 2023-12-19:
Hmm, interesting. It works on my side, as shown in the following screenshots:

[image]
[image]

Guanbin-Huang, 2023-12-19:
you can start to download it. but it ends up with around 92%, then it pops up the error.

yufeiwang63, 2023-12-19:
The downloading finished correctly on my side. Could it be a network issue?
[image]

Guanbin-Huang, 2023-12-19:
oh, maybe~,thanks

closed 2023-12-19 by Guanbin-Huang