Genesis-Embodied-AI/RoboGen #15: visualization

Guanbin-Huang, 2023-12-19:
it seems like there's nothing to visualize?

I only get the folloiwng
[image]

how to get the visuzlaition like what you did in the paper?
[image]

Guanbin-Huang, 2023-12-19:
i run  python run.py

> it seems like there's nothing to visualize?
> 
> I only get the folloiwng ![image](https://private-user-images.githubusercontent.com/30463291/291485782-383a034c-e0ed-4834-b502-49fed7418e9e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI5NjQ1MTYsIm5iZiI6MTcwMjk2NDIxNiwicGF0aCI6Ii8zMDQ2MzI5MS8yOTE0ODU3ODItMzgzYTAzNGMtZTBlZC00ODM0LWI1MDItNDlmZWQ3NDE4ZTllLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE5VDA1MzY1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJkNzVjNTJlYmNiZDI3ODA4MmVkZTYyYjI1ZTY2MzMyNTc3NzNjZjM0YTI1YWIxMzQ1M2QzMWVmODAyZTRlM2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.j2bociPeFJ1EGf_ipVBr8Kk5TSOhoLN0OttB85lIfXI)
> 
> how to get the visuzlaition like what you did in the paper? <img alt="image" width="743" src="https://private-user-images.githubusercontent.com/30463291/291485969-0f45f18c-83cd-4361-a589-5095f549567d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI5NjQ1MTYsIm5iZiI6MTcwMjk2NDIxNiwicGF0aCI6Ii8zMDQ2MzI5MS8yOTE0ODU5NjktMGY0NWYxOGMtODNjZC00MzYxLWE1ODktNTA5NWY1NDk1NjdkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjE5VDA1MzY1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFlYjBiMzE4MmUxZjBmY2RkNjljMmU4NGNlODY3ZTRiYTIzMWQ1YzI3MGQ5MTQyYzUwMzRhMDcyNGRjMGQ5NmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.SLabmhuwtKuiVW98biFFS9aRoE2OylpqI92abvvTBsw">

yufeiwang63, 2023-12-19:
The photo-realistic rendering as shown in the paper is only supported using Genesis. This implementation in bullet uses bullet's default rendering and the rendering quality will be lower.

running `python run.py` will first randomly generate some tasks, which is the output from your terminal, and then try to solve the task, which will then produce some gifs of the robot doing the task. 

If you want to quickly see the robot working, you can try skip the task generation and directly learn some pre-generated tasks, by running `python execute_locomotion.py --task_config_path example_tasks/task_Turn_right/Turn_right.yaml` or `python execute.py --task_config_path example_tasks/Change_Lamp_Direction/Change_Lamp_Direction_The_robotic_arm_will_alter_the_lamps_light_direction_by_manipulating_the_lamps_head.yaml`. It should take roughly ~10-20 minutes to solve the locomotion task with CEM, and 3 hours to solve the manipulation task with RL. 

See https://github.com/Genesis-Embodied-AI/RoboGen/issues/7#issuecomment-1804924329 about where the visuals will be stored.

closed 2023-12-26 by yufeiwang63