Genesis-Embodied-AI/RoboGen #16: link_id is none

Guanbin-Huang, 2023-12-25:
command i run

python execute.py --task_config_path data/generated_task_from_description/Put_a_pen_into_the_box_box_100426_2023-12-25-11-43-33/Put_a_pen_into_the_box_The_robot_arm_picks_up_a_pen_and_places_it_inside_the_box_by_opening_the_lid_of_the_box_putting_the_pen_inside_and_then_closing_the_lid.yaml

the put_a_pen_into_box is the demo for generation  in your readme.

the yaml file
`-   use_table: true
-   all_uid:
    - 69d4ff6819d74b98bc2934f389d4ba79
    - 720f8512d54f42abb54ee389c06db182
    - f43199f19c3142c68cc672db55d9a40d
    - d31aa1b82956407585c62ea7b5810057
    - 5aec17305b0d4d0698b72cb1968e3a60
    - 6d1fc6b75f524b0aa34861c7e4a3ce05
    center: (0.3, 0.6, 0)
    lang: a common box with a lid
    name: Box
    on_table: true
    path: box.urdf
    size: 0.3
    type: mesh
    uid:
    - 69d4ff6819d74b98bc2934f389d4ba79
    - 720f8512d54f42abb54ee389c06db182
    - f43199f19c3142c68cc672db55d9a40d
    - d31aa1b82956407585c62ea7b5810057
    - 5aec17305b0d4d0698b72cb1968e3a60
    - 6d1fc6b75f524b0aa34861c7e4a3ce05
-   all_uid:
    - b60e5ed77ce54248a44d40fc69feb60e
    - 42e02b8cb8884ce09380943e8dd90372
    - be6e3308c49841778c4de0cabf36432e
    - 4db2723862d8402daa690f97e6cbd2ef
    - 100aba3dbb124819aded5be4ecbf978f
    - 18b147a4dca547649b5f6ee65c954458
    center: (0.7, 0.2, 0)
    lang: a standard pen
    name: Pen
    on_table: true
    path: pen.obj
    size: 0.01
    type: mesh
    uid:
    - b60e5ed77ce54248a44d40fc69feb60e
    - be6e3308c49841778c4de0cabf36432e
    - 4db2723862d8402daa690f97e6cbd2ef
    - 100aba3dbb124819aded5be4ecbf978f
-   solution_path: data/generated_task_from_description/Put_a_pen_into_the_box_box_100426_2023-12-25-11-43-33/task_Put_a_pen_into_the_box
-   set_joint_angle_object_name: box
-   spatial_relationships: []
-   task_description: The robot arm picks up a pen and places it inside the box by
        opening the lid of the box, putting the pen inside, and then closing the lid
    task_name: Put a pen into the box
-   distractor_config_path: data/generated_task_from_description/Put_a_pen_into_the_box_box_100426_2023-12-25-11-43-33/The_robot_arm_picks_up_a_pen_and_places_it_inside_the_box_by_opening_the_lid_of_the_box_putting_the_pen_inside_and_then_closing_the_lid_distractor.yaml
`

[image]

Guanbin-Huang, 2023-12-25:
can you give us a docker to run it? Otherwise it's torturing. 
Any additional info you would like to add for? please tell me, thanks.

Guanbin-Huang, 2023-12-25:
[image]
i've got the similar issue in the case of "cup"
"cup" and "box", both of them don't have the link_id, which are really confusing.

yufeiwang63, 2023-12-25:
Thanks for the question. 

It seems that the generated box object in the generated yaml config file is having type 'mesh' instead of 'urdf'. The latter should be the correct one since 'urdf' means it is an articulated object, which means it will have joints and links. The 'mesh' type of object is non-articulated, and will have no joints and links. 

I think the error should be that when you run the generation command, you didn't capitalize the first letter of "box". I.e., you should run
`python gpt_4/prompts/prompt_from_description.py --task_description "Put a pen into the box" --object "Box"`
instead of 
`python gpt_4/prompts/prompt_from_description.py --task_description "Put a pen into the box" --object "box"`
The reason is that only the objects in PartNetMobility will be articulated, so you need to stick to their conventions. ”Box“ is a proper articulated object category in PartNetMobility yet 'box' is not. You can check the full list of supported articulated object categories in this file: https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/data/partnet_mobility_dict.json#L1703. As you can see, they use the capitalized word "Box". 

I have added a check for making sure the articulated object is supported by partnetmobility in the new commit. 

Let me know if you have any more questions.

yufeiwang63, 2023-12-25:
Also I noticed that in the generated config, the pen is somewhat very small -- only 0.01m. This sometimes happens when GPT somewhat gives very wrong object sizes. You might want to correct that as well or re-generate, or update the prompt. The size prompt we use is here: https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/gpt_4/prompts/prompt_with_scale.py

Guanbin-Huang, 2023-12-26:
hi, what if i use multiple object? let's say i don't just need Box but also coffeepot?

yufeiwang63, 2023-12-26:
Right now the task generation needs/samples 1 articulated object from PartNetMobility as the core object for the task, and depends on the generated task, there might be other objects involved. E.g., for the task of putting pen to the box, you only need to specify the Box as the core object, and the pen will be automatically built/retrieved by RoboGen.

What kind of tasks do you have in mind that require multiple objects?

Guanbin-Huang, 2023-12-27:
Let's talk about the phrase "put the pen into a box". As you can see, the phrase involves two items, "pen" and "box". But why do we only focus on the box instead of the two objects?

zhouxian, 2023-12-27:
RoboGen is about generating diverse tasks and object sampling functions as
the seeding stage for increase the diversity of the auto generated tasks.
The task you described, could be generated by the task proposal stage and
the corresponding items will be retrieved. Following specific user-provided
task prompt is not a focus at the current stage.

Also, it’s somewhat confusing to me why you have been shooting questions in
such a condescending and aggressive manner - you could clearly use a bit
more politeness.

Guanbin-Huang, 2023-12-28:
> RoboGen is about generating diverse tasks and object sampling functions as the seeding stage for increase the diversity of the auto generated tasks. The task you described, could be generated by the [...]

sorry~~, thank you for your pointing out my manner. I didn't mean it. Your confusion is correct.  Just talk to gpu in a teacher-student manner, i copied paste it without second thinking.

i wish it wouldn't bother you too much.

closed 2023-12-28 by Guanbin-Huang

zhouxian, 2023-12-28:
Thank you for your reply. No worries :) Do let us know if you've got any more questions