Genesis-Embodied-AI/RoboGen #20: Python OMPL Module issues on Windows

Captain-Freels, 2024-02-29:
I am having an issue getting the OMPL GUI to work on my computer. When I run 'ompl_app.py' from the software package, I get this error:

File "C:\vcpkg\buildtrees\omplapp\src\1.5.1-6c8b6a14e1.clean\gui\ompl_app.py", line 56, in
   from ompl.util import OutputHandler, useOutputHandler, LogLevel,
ModuleNotFoundError: No module named 'ompl'

I tried using pip to install it but it keeps saying it doesn't exist. What can I do to mend this?

yufeiwang63, 2024-03-01:
There is probably some error in the OMPL installation. 
BTW, we are not using ompl_app.py for this project.

Captain-Freels, 2024-03-04:
When I revisit the installation instructions, it says vcpkg does not contain the python bindings. What can I do to get said bindings or is there another way to run the GUI?

yufeiwang63, 2024-03-09:
In this project we are using just the python bindings, not the GUI itself. So it won't be necessary to install the GUI if you want to use RoboGen.

yufeiwang63, 2024-03-20:
Hi,

If you still have trouble installing ompl python binding from source, you can try installing it from the precompiled wheels, see the updated ReadMe for details.

closed 2024-03-27 by yufeiwang63