Genesis-Embodied-AI/RoboGen #21: AttributeError:'NoneType' object has no attribute 'dumps'

nnynngn, 2024-03-13:
hi, thank you for your impressive work.
when I run ‘’python execute_locomotion.py --task_config_path example_tasks/task_Turn_right/Turn_right.yaml‘’,
the results are following:

Exception ignored in: <function Pool.__del _ at x7f7adf5d2550>Traceback(most recent call last):
File "/home/anaconda3/envs/robogen/lib/python3.9/multiprocessing/pool.py", line 268, in __del__
File " /home/anaconda3/envs/robogen/lib/python3.9/multiprocessing/queues.py", line 371, in put
AttributeError:'NoneType' object has no attribute 'dumps'

Could you tell me how to solve it?

yufeiwang63, 2024-03-14:
Hi, 

Can you check [this](https://github.com/Genesis-Embodied-AI/RoboGen/issues/15#issuecomment-1863405119) to see if a mp4 file of the learned behavior has been generated?

If the mp4 file is successfully generated, this error can be safely ignored for now; this is probably due to sth internally wrong with the python multiprocessing library. 

Thanks for reporting this, I will update the ReadMe to mention this as well.

yufeiwang63, 2024-03-20:
Hi, 

If you still have trouble installing ompl python binding from source, you can try installing it from the precompiled wheels, see the updated ReadMe for details.

closed 2024-03-21 by yufeiwang63