Genesis-Embodied-AI/RoboGen #22: when trying to learn skill has ERROR

wfy1126, 2024-03-27:
trying to learn skill:  data/generated_tasks_release/Toilet_102684_2024-03-27-10-50-37/Flush_Toilet_The_robotic_arm_will_actuate_the_lever_to_flush_the_toilet.yaml
task_config_path:  data/generated_tasks_release/Toilet_102684_2024-03-27-10-50-37/Flush_Toilet_The_robotic_arm_will_actuate_the_lever_to_flush_the_toilet.yaml
pybullet build time: Nov 28 2023 23:52:03
all substeps:
 
all substep types:
 
all action spaces:
 
Traceback (most recent call last):
  File "/media/wfy/U盘/Robogen/RoboGen/execute.py", line 272, in <module>
    execute(task_config_path, resume=args.resume, training_algo=args.training_algo, time_string=args.time_string, 
  File "/media/wfy/U盘/Robogen/RoboGen/execute.py", line 244, in execute
    save_numpy_as_gif(np.array(all_rgbs), "{}/{}-{}.gif".format(save_path, "all", time_string))
  File "/media/wfy/U盘/Robogen/RoboGen/manipulation/utils.py", line 212, in save_numpy_as_gif
    clip = ImageSequenceClip(list(array), fps=fps).resize(scale)
  File "/home/wfy/anaconda3/envs/robogen/lib/python3.9/site-packages/moviepy/video/io/ImageSequenceClip.py", line 64, in __init__
    if isinstance(sequence[0], str):
IndexError: list index out of range

closed 2024-04-01 by wfy1126

huiyiygy, 2024-11-05:
Hi，I have the same error when I run the command 'python run.py' 
Did you solve it?

wfy1126, 2024-11-05:
> Hi，I have the same error when I run the command 'python run.py' Did you solve it?

不太记得具体怎么解决的了，好像是跟GPT回答的语句格式有关。你可以看看对GPT回答的处理

huiyiygy, 2024-11-06:
> > Hi，I have the same error when I run the command 'python run.py' Did you solve it?
> 
> 不太记得具体怎么解决的了，好像是跟GPT回答的语句格式有关。你可以看看对GPT回答的处理

感谢提示，的确是GPT回答的语句格式发生了改变:)
在gpt_4/prompts/prompt_manipulation_reward_primitive.py文件中，原本解析回答时，需要匹配以substep开头的行
```python
line.startswith("substep")
```
但返回的结果里变成了### Substep。使得每个动作解析后保存的substeps.txt、substep_types.txt、action_spaces.txt均为空，最终触发上面这个报错。改成下面这种就可以了
```python
line.startswith("### substep")
```
另外，一开始我调用的是gpt4o-mini，GPT回答的格式不固定，也会导致匹配失败。换成gpt4o后，回答的格式就很稳定了