Genesis-Embodied-AI/RoboGen #23: Can I import different robots for use in this project?

wfy1126, 2024-04-01:
I want to replace the robots in the project, but I don't know how to do it

yufeiwang63, 2024-04-01:
You should be able to chnage the robots by using different urdf files of the robot. See this issue for more details: https://github.com/Genesis-Embodied-AI/RoboGen/issues/2

wfy1126, 2024-04-02:
Thanks for your kind replies!

closed 2024-04-04 by yufeiwang63

wfy1126, 2024-04-29:
Hello, I noticed in your code that robots performing operational tasks rely on sliding for movement. I have a wheeled robot and I want it to move based on the rotation of the wheels. What modifications should I make. 

Looking forward to your reply.
Hello, I noticed in your code that robots performing operational tasks rely on sliding for movement. I have a wheeled robot and I want it to move based on the rotation of the wheels. What modifications should I make. Looking forward to your reply.

wufeiyang
***@***.***