Genesis-Embodied-AI/RoboGen #27: Segmentation error from "solved = self.ss.solve(allowed_time)"

ubgaokk, 2024-05-26:
When I am running ```python execute.py --task_config_path example_tasks/Change_Lamp_Direction/Change_Lamp_Direction_The_robotic_arm_will_alter_the_lamps_light_direction_by_manipulating_the_lamps_head.yaml```, I got the segmentation error as following.  I found the problem is coming from
```
solved = self.ss.solve(allowed_time)
```
Could you help me understand why? Thanks.
```
...
 99% [ Simplify convex-hulls   0% ] Simplify convex-hulls   0%
100% [ Simplify convex-hulls 100% ] Simplify convex-hulls 100%
object a desktop computer monitor choosing uid d6c02c88cf574f9db8be40761a070b45 urdf_path objaverse_utils/data/obj/d6c02c88cf574f9db8be40761a070b45/material_non_vhacd.urdf
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
baseb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_0b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_1b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_2b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_3b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_4b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
baseb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_0b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_1b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_2b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_3b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_4Finished loading object:  lamp
Finished loading object:  notebook
b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
baseb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_0b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_1b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
baseb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_0b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:
link_1Finished loading object:  pen
Finished loading object:  coffee mug
Finished loading object:  computer monitor
WARNING - 2024-05-25 19:25:00,815 - warnings - /home/kanke/miniconda3/envs/robogen/lib/python3.9/site-packages/gym/spaces/box.py:73: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(

pbompl joint number:  10
pbompl joint idx:  [0, 1, 2, 4, 5, 6, 7, 8, 9, 10]
Joint bounds: [[-2.0, 2.0], [-2.0, 2.0], [-5.0, 5.0], [-2.9671, 2.9671], [-1.8326, 1.8326], [-2.9671, 2.9671], [-3.1416, 0.0], [-2.9671, 2.9671], [-0.0873, 3.8223], [-2.9671, 2.9671]]
motion planning obstables:  [0, 2, 3, 4, 5, 6, 7]
robot allow collision links:  []
ik_lower_limits:  [-2.     -2.     -5.     -2.9671 -1.8326 -2.9671 -3.1416 -2.9671 -0.0873
 -2.9671]
ik_upper_limits:  [2.     2.     5.     2.9671 1.8326 2.9671 0.     2.9671 3.8223 2.9671]
sampling target ik it:  0
sampling target ik it:  10
sampling target ik it:  20
sampling target ik it:  30
joint:  0  lower limit:  -2.0  upper limit:  2.0  target:  -1.367744866674962
joint:  1  lower limit:  -2.0  upper limit:  2.0  target:  -0.2514780671486595
joint:  2  lower limit:  -5.0  upper limit:  5.0  target:  3.914049405321227
joint:  3  lower limit:  -2.9671  upper limit:  2.9671  target:  0.4804371549084728
joint:  4  lower limit:  -1.8326  upper limit:  1.8326  target:  1.089252595025869
joint:  5  lower limit:  -2.9671  upper limit:  2.9671  target:  0.836866881952676
joint:  6  lower limit:  -3.1416  upper limit:  0.0  target:  -0.923762920972401
joint:  7  lower limit:  -2.9671  upper limit:  2.9671  target:  0.6614467777102168
joint:  8  lower limit:  -0.0873  upper limit:  3.8223  target:  1.98288515326969
joint:  9  lower limit:  -2.9671  upper limit:  2.9671  target:  -0.9102207439824311
start_planning
Warning: kBITstar: This option no longer has any effect; Rewiring is never delayed.
         at line 1261 in /tmp/binarydeb/ros-noetic-ompl-1.5.2/src/ompl/geometric/planners/informedtrees/src/BITstar.cpp
delay_rewiring_to_first_solution = 0

Segmentation fault (core dumped)
```

yufeiwang63, 2024-05-31:
Hi,

I think this indicates that there is some error in the installed ompl python binding.
Can you provide a bit more details on how you installed it (e.g., did you compile it yourself, or installed the pre-released version from pip), and your system information?

ubgaokk, 2024-06-01:
I have installed the pre-released version of ompl
pip install ompl-1.6.0-cp39-cp39-manylinux_2_28_x86_64.whl
My system is Ubuntu 20.04.6 LTS.
It showed the segmentation error as above.
```
NVIDIA-SMI 545.23.08              Driver Version: 545.23.08    CUDA Version: 12.3
```

yilin404, 2024-10-28:
@ubgaokk I’m experiencing the same issue. Have you found any solutions? I’d really appreciate any insights you might have. Thanks!

ubgaokk, 2024-10-29:
@yilin404 I have got some solution: 
It includes two steps (draft note I wrote before)
1.Fully compile ompl
https://github.com/ompl/ompl
Use the commit from main
commit 8c9512f7339c0db63c54118e21f7dd2a7bb320d2
Under no virtual env, compile it by running following
cmake ../.. -DPYTHON_EXEC=/usr/bin/python3
make update_bindings -j 32
make -j 32

Then (this determines which ompl will be called)
echo "/home/xxxx/Downloads/test_ompl/ompl/py-bindings" >> ~/miniconda3/envs/ompl/lib/python3.9/site-packages/ompl.pth

2. Compile ompl_util
-first run the
cmake ../.. -DPYTHON_EXEC=/home/xxxx/miniconda3/envs/ompl/bin/python3.9
make update_util_bindings
-then
cmake ../.. -DPYTHON_EXEC=/usr/bin/python3
Using ```make update_util_bindings``` to generate _util.so instead of ```make update_bindings```

It should be OK for you to run '''execute.py'''.

yilin404, 2024-10-30:
Thank you for your response and suggestions! I will try the method you recommended and provide feedback on the results. I really appreciate your help!

yilin404, 2024-10-30:
@ubgaokk  Thank you for your response and suggestions! I will try the method you recommended and provide feedback on the results. I really appreciate your help!

sunxin010205, 2024-11-25:
> @ubgaokk Thank you for your response and suggestions! I will try the method you recommended and provide feedback on the results. I really appreciate your help!

Hello, I would like to ask if you have solved this problem？