Genesis-Embodied-AI/RoboGen #29: render the scenes in high quality

WSBG, 2024-09-04:
Thanks for sharing your wonderful work!
After completing my training with PyBullet, I’m interested in rendering the scenes to match the high visual quality seen in your demo. 
Unfortunately, we might not have the time to wait for the Genesis release.
Could you please give me some advice or guidance on how to achieve that level of rendering?

yufeiwang63, 2024-09-06:
Hi,

This repo might be useful: https://github.com/huy-ha/pybullet-blender-recorder
It exports a scene from pybullet and then renders it in blender.

WSBG, 2024-09-06:
Thank you for your suggestion！

WSBG, 2024-10-02:
> Hi,
> 
> This repo might be useful: https://github.com/huy-ha/pybullet-blender-recorder It exports a scene from pybullet and then renders it in blender.
However, it seems that this codebase does not provide me with a method to import the robot arm and URDF files? Especially for the robot arm, I have no idea how to import it QAQ.

kingki663, 2025-01-03:
> Hi,
> This repo might be useful: https://github.com/huy-ha/pybullet-blender-recorder It exports a scene from pybullet and then renders it in blender.
> However, it seems that this codebase does not [...]

Genisis is available now：https://github.com/Genesis-Embodied-AI/Genesis

WSBG, 2025-01-05:
> Genisis is available now：https://github.com/Genesis-Embodied-AI/Genesis

Thank you!

closed 2025-01-05 by WSBG