Genesis-Embodied-AI/RoboGen #30: KeyError: 'a common soda can_all'

WSBG, 2024-09-05:
Hi,

Thank you for the great work on this project. After successfully running the code, I decided to personalize the generated tasks by modifying the prompts. Specifically, I altered the prompt text and the four provided examples in the prompt_manipulation.py file. However, this led to an error.

Could you please provide guidance on how to resolve this issue?

I appreciate your help!

`pybullet build time: Nov 28 2023 23:52:03
Traceback (most recent call last):
  File "/scratch4/danielk/lshen30/torch/robo/RoboGen/execute_long_horizon.py", line 179, in <module>
    execute_multiple_try(task_config_path, 
  File "/scratch4/danielk/lshen30/torch/robo/RoboGen/execute_long_horizon.py", line 102, in execute_multiple_try
    rgbs, states = execute_primitive(task_config_path, solution_path, substep, last_restore_state_file, save_path, 
  File "/scratch4/danielk/lshen30/torch/robo/RoboGen/execute.py", line 14, in execute_primitive
    env, safe_config = build_up_env(task_config, solution_path, task_name, last_restore_state_file, 
  File "/scratch4/danielk/lshen30/torch/robo/RoboGen/manipulation/utils.py", line 157, in build_up_env
    env = env_class(**save_config)
  File "/scratch4/danielk/lshen30/torch/robo/RoboGen/data/generated_tasks_release/TrashCan_102254_2024-09-04-04-09-28/task_Dispose_of_Waste/grasp_the_piece_of_waste.py", line 11, in __init__
    super().__init__(*args, **kwargs)
  File "/scratch4/danielk/lshen30/torch/robo/RoboGen/manipulation/sim.py", line 80, in __init__
    self.set_scene()
  File "/scratch4/danielk/lshen30/torch/robo/RoboGen/manipulation/sim.py", line 226, in set_scene
    use_table, articulated_init_joint_angles, spatial_relationships = self.load_and_parse_config(restore_state)
  File "/scratch4/danielk/lshen30/torch/robo/RoboGen/manipulation/sim.py", line 331, in load_and_parse_config
    res = download_and_parse_objavarse_obj_from_yaml_config(distractor_config_path, candidate_num=self.object_candidate_num, vhacd=self.vhacd)
  File "/scratch4/danielk/lshen30/torch/robo/RoboGen/manipulation/utils.py", line 125, in download_and_parse_objavarse_obj_from_yaml_config
    obj['all_uid'] = text_to_uid_dict[obj["lang"] + "_all"]
KeyError: 'a common soda can_all'
`

WSBG, 2024-09-06:
I have observed that certain objects within the objaverse_utils/text_to_uid.json file are accompanied by additional keys; for instance, the key "a bowl of tomato soup" is followed by "a bowl of tomato soup_all". However, this does not seem to be the case for all objects, as indicated by the KeyError: 'a common soda can_all' that I've encountered. In the manipulation/utils.py file, specifically on line 125, there is a requirement to retrieve obj['all_uid'] = text_to_uid_dict[obj["lang"] + "_all"]. This inconsistency appears to be the cause of the aforementioned error. Could you kindly provide some guidance on how to address this issue? I would greatly appreciate your assistance. Thank you for your time and consideration.
[image]
No "a common soda can_all"

huiyiygy, 2024-12-04:
把text_to_uid.json中已有的a common soda can去掉。重新跑，代码检索不到，就会重新去检索生成a common soda can 和a common soda can_all了

WSBG, 2024-12-10:
> o

谢谢

closed 2024-12-10 by WSBG