Genesis-Embodied-AI/RoboGen #37: Soft body manipulation tasks

ccxi1008, 2025-02-07:
Hi, I've learned that you havn't include code of this part since soft body dynamics is not supported in pybullet, but I'm still wondering how to insert the exsisting text-image-3d pipeline into this work and how can I get the visual results like yours, is there any relevant codes(which I think it's very different from this reproducing codebase)?
Another question is that I want to use text-image-3d pipeline to directly generate the additional objects in the scene instead of retrieving from Objaverse, how can I modify the code? Must I have to pre-genarate all impossible meshes and choose the correlated ones every time?
Thanks very much!!!

yufeiwang63, 2025-02-12:
Hi,

Thanks for your interest in our work & code.

Back then, we used this repo for text-to-image-to-mesh generation:  https://github.com/ashawkey/stable-dreamfusion for image-2-3d generation.

But, I believe the text23d generation field is evolving rapidly so there should be many more new repos these days that can do much better & faster generation.

Yes, I think a good way to modify the existing code to incorporate the generated objects could be:
1) pre-generate the objects offline and just choose between them online
2) you can generate the objects online given the returned text descriptions of the objects. I.e., at this line of code: https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/utils.py#L119, you can choose to generate the object mesh instead of retrieving it from Objaverse.

ccxi1008, 2025-02-12:
> Hi,
> 
> Thanks for your interest in our work & code.
> 
> Back then, we used this repo: https://github.com/ashawkey/stable-dreamfusion for image-2-3d generation.
> 
> But, I believe the text23d generation field is evolving rapidly so there should be many more new repos these days that can do much better & faster generation.
> 
> Yes, I think a good way to modify the existing code to incorporate the generated objects could be:
> 
> 1. pre-generate the objects offline and just choose between them online
> 2. you can generate the objects online given the returned text descriptions of the objects. I.e., at this line of code: https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/utils.py#L119, you can choose to generate the object mesh instead of retrieving it from Objaverse.

Thanks a lot! But I'm still confused if I want to pre-generate the objects **offline**, where can I directly give/choose  the corresponding mesh path which including the description of objects for the system to use? Since I've found all the functions and classes are structed with id(uid) input, it seems hard to modify all the codes using id(uid）.

closed 2025-03-09 by yufeiwang63