Genesis-Embodied-AI/RoboGen #39: GIF / Visualization output seems wrong？

ccxi1008, 2025-02-13:
I‘ve learned from #15 #29 that if I want to get the results like your paper's, I need to use Genesis or other tools.
But before that , I run run.py with some example_tasks, and got gif results like these(with some offline mesh inserted):

[image]

[image]

[image]

Is it true to get these all nearly **white** objects so far by finish run.py?(even selected urdf from PartNet or already add my albedo image when using obj_to_urdf) And the scenes don't have true walls and floors……
Where are the problem and what should I do? After that, how can I get the results like your paper's (maybe how to insert which result file to Genesis)?
Sorry to bother you again but thanks very much!

yufeiwang63, 2025-02-20:
Hi,

The "white" of the objects is due to the simple renderer used in pybullet. Pybullet has trouble correctly loading and rendersing some of the object textures.

Regarding walls and floors -- for the gifs on the website, we just loaded a floor with textures to improve the rendering quality when renderered in Genesis. 

If rendering is a strong need, I can write a small script that loads all the stored pybullet states (objects and robots) into Genesis and you can do the rendering (e..g, using the photo-realistic Luisa renderer) there to get better looking results.

Thanks,
Yufei

yufeiwang63, 2025-02-24:
Hi,

The "center" parameter works as follows:
- For objects that are on the table, the coordinate is expressed relative to the table coordinate frame, i.e., (0, 0) means the lower corner of the table and (1, 1) means the upper corner of the table. (0.5, 0.5) means the center of the table.
- For objects that are not on the table, the coordinate frame is in the world frame.

There are some addition post-processing on the "center" parameter:
- We adjust the height of the object so the lowest point on the object aligns with the height specified by GPT: https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/sim.py#L235-L236
- We resolve collisions between objects: e.g., if two objects are too close that they collide with each other, we push them away: https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/sim.py#L238-L239
- I think we also did this post-processing where we push objects that too close to the robot itself further away: https://github.com/Genesis-Embodied-AI/RoboGen/blob/main/manipulation/sim.py#L156

These post-process may result in the case that the object is not exactly at the location where you specified in the "center" parameter. You could disable some of them if you really want the object to stay where your input is.